nebbian/RoboxSlicerExtension #3: project structure

natdan, 2017-05-30:
@nebbian & @Benraay - do you want, maybe, me creating java project structure for you guys? Any ideas what should be project's groupId and artifactId? Package name? I know those are the least important things and the hardest for me to come up with something sensible! O: )

Benraay, 2017-05-30:
Go for it !
So I can start the translation to Java !

-- 

Benjamin Raaymakers

CONFIDENTIALITY NOTICE:

The contents of this email message and any attachments are intended solely
for the addressee(s) and may contain confidential and/or privileged
information and may be legally protected from disclosure. If you are not
the intended recipient of this message or their agent, or if this message has
been addressed to you in error, please immediately alert the sender by
reply email and thendelete this message and any attachments. If you are not
the intended recipient, you are hereby notified that any use,
dissemination, copying, or storage of this message or its attachments
is strictly
prohibited.

nebbian, 2017-05-30:
Doing the framework for this project and planning it out well, would be fantastic.  I'd suggest the package name might be RoboxSlicerExtension, but I don't know much about the normal conventions on this sort of thing.  The last time I touched Java was at university, so that's... ummm... 15 years ago... 

You'd be a lot better placed to make those sorts of decisions.

I think we need to decide on the scope of what we want this project to do.  My $0.02 is that it should do this:

- Easily install and uninstall itself (you've already done this)
- Allow easy selection of slicer (Slic3r, Cura, Simplify3D, etc)
- Integrate as much with AutoMaker as possible.  This includes grabbing the config file that AutoMaker creates, parsing it, and feeding those values to the chosen slicer.  It should also respect the material choices that the user makes inside AutoMaker.
- Show a preview, if the existing slicer doesn't already have one.  This might be an option for if people are happy to use the inbuilt Cura Robox slicer.  (you've already done the work for this I think)
- Spit out valid GCode to the Robox Post Processor, no matter which slicer is chosen.

I can't think of anything else, if this is the list then we need to think of a good name for something that does all those things.  AutoMakerHelper?  RoboxSlicerExtension?

natdan, 2017-05-30:
:+1:

Project name goes without saying: 'Robox Slicer Extension' ('robox-slicer-extension' for maven artifactId, 'roboxslicerextension' for last part of the package, RoboxSlicerExtension for the class name, etc)

Group - that's where the problem is. Maven groupId, first part of the package (sometimes last part of the domain name) is where you describe 'organisation'/'group'/'group of projects', etc... The worse would be to say 'org.githib' as we have nothing to do with github except it being hosted here. Other options are 'com.roboxslicerextension', 'net.roboxslicerextension'... Any other 'virtual' company/organisation/something-else would be great! I have my own projects always done under org.ah (for short) or org.abstracthorizon as I have create web site (in days before github and such) and posted my projects there. I still do that ([http://www.abstracthorizon.org](http://www.abstracthorizon.org) and many other subdomains)

"Show a preview, if the existing slicer doesn't already have one. This might be an option for if people are happy to use the inbuilt Cura Robox slicer. (you've already done the work for this I think)"
If we set up separate executable (of some kind) where you can select which flow do go with (which slicer to have) I would use the same app to select if such preview (see here: [https://github.com/natdan/gcode-preview](https://github.com/natdan/gcode-preview)) can be switched on or off from the 'flow'. Also, same app should be able to set up any other local scripts user wants to add (some that just modify 'raw' gcode somehow/set pause gcode at particular level, etc).

👍 for all the other bullet points you put! Plus - try to keep it as platform agnostic as possible (read: it should work on Windows, as well! :))

nebbian, 2017-05-30:
The group could be something like org.roboxenthusiasts?  org.roboxslicerextension works as well.
+1 for making it cross platform.  For some reason us Mac people are in the majority for once, I don't want to treat Windows/Linux people with the same disdain that they've treated us with for years.

natdan, 2017-05-30:
_"For some reason us Mac people are in the majority for once, I don't want to treat Windows/Linux people with the same disdain that they've treated us with for years."_
LOL. I'm happy with org.roboxenthusiasts unless someone has better idea/strong(er) opinion(s)! :)

natdan, 2017-05-30:
Oh, oh... There's another option (but not that good): com.roboxing! It is Tom Gidden's original web site ([https://github.com/tomgidden](https://github.com/tomgidden)) currently hosted at my servers... So it can accommodate (if really necessary) 'web side' of the project (no matter of name of the 'group').

nebbian, 2017-05-30:
com.roboxing is probably the best one yet.  Nice catch!

Benraay, 2017-05-31:
go for the last option !
com.roboxing

Benjamin Raaymakers

CONFIDENTIALITY NOTICE:

The contents of this email message and any attachments are intended solely
for the addressee(s) and may contain confidential and/or privileged
information and may be legally protected from disclosure. If you are not
the intended recipient of this message or their agent, or if this message has
been addressed to you in error, please immediately alert the sender by
reply email and thendelete this message and any attachments. If you are not
the intended recipient, you are hereby notified that any use,
dissemination, copying, or storage of this message or its attachments
is strictly
prohibited.

natdan, 2017-05-31:
LOL - already in the pull request! :)

Benraay, 2017-05-31:
sorry I have a lot of work this week my replies are a bit late you make
good choices with @nebbian I let you decide for the first project structure
!

Benjamin Raaymakers

CONFIDENTIALITY NOTICE:

The contents of this email message and any attachments are intended solely
for the addressee(s) and may contain confidential and/or privileged
information and may be legally protected from disclosure. If you are not
the intended recipient of this message or their agent, or if this message has
been addressed to you in error, please immediately alert the sender by
reply email and thendelete this message and any attachments. If you are not
the intended recipient, you are hereby notified that any use,
dissemination, copying, or storage of this message or its attachments
is strictly
prohibited.

natdan, 2017-05-31:
Oh, no - don't let you stop expressing what you had in mind. Nothing is set in stone!

nebbian, 2017-06-30:
Thanks for the discussion guys.  I think we've ended up with a beauty!

closed 2017-06-30 by nebbian