nebbian/RoboxSlicerExtension #7: Compiling Java Code

natdan, 2017-06-01:
> I also got stuck compiling the java source in Jar I did it with command line also but when I try to launch it I have a "no main or something" error How do you compile the jar ? Should be easy as there is only one class. My pure java tests are a bit far from me about 7 years But I merged the master update to the safetomerge branch and cause nebbia

Get the latest maven ([https://maven.apache.org/download.cgi](https://maven.apache.org/download.cgi)).

From command line go to java/robox-slicer-extension and type:
```
mvn clean install
```
When it finishes in java/robox-slicer-extension/robox-slicer-control/target and java/robox-slicer-extension/robox-slicer-flow/target you can find two jar files like: 
```
original-robox-slicer-control-1.0-SNAPSHOT.jar
robox-slicer-control-1.0-SNAPSHOT.jar
```
The bottom one is 'executable' jar. In IDE's you don't need any of it. Just import it as maven project (from the top: java/robox-slicer-extension) find 'Main' class in both projects (robox-slicer-control and robox-slicer-flow) and run that class. IDE will do the rest of setting up classpath, etc... 

Let me know if you stumble at any point - I'm more than happy to help you set it up!

Benraay, 2017-06-01:
Great I will try to reimport as maven project in intelliJ.
If I got stuck I will use command line thanks !!

Benjamin Raaymakers

CONFIDENTIALITY NOTICE:

The contents of this email message and any attachments are intended solely
for the addressee(s) and may contain confidential and/or privileged
information and may be legally protected from disclosure. If you are not
the intended recipient of this message or their agent, or if this message has
been addressed to you in error, please immediately alert the sender by
reply email and thendelete this message and any attachments. If you are not
the intended recipient, you are hereby notified that any use,
dissemination, copying, or storage of this message or its attachments
is strictly
prohibited.

natdan, 2017-06-01:
No need to command line unless you want to produce standalone - executable jar. In IntelliJ just navigate to Main class (of each project) and use 'Run...' on it!

Benraay, 2017-06-02:
Ok I got it compiled.
I wanted to start writing the postProcessor but I saw that you @natdan started to write some code how can I get your work before I start mine, do I need to  merge your feature branch to mine ? Or need to pull your work on @nebbian's repo first ?
Also where do It write the postProcessor, do I make a new Class ? it's maybe the best way of doing.

natdan, 2017-06-03:
I've started it last night. I need another day and I'll make a pull request. It is, currently, just a re-write of shell script + perl. I've made work-in-progress commit so you can see where I am going with it. I hope I'll get an hour tonight and maybe couple tomorrow to get it to state you can start adding post processor code to slicer(s). Check it in here: [https://github.com/natdan/RoboxSlicerExtension/tree/Initial-Flow-Code/java/robox-slicer-extension/robox-slicer-flow/src/main/java/com/roboxing/slicerextension/flow](https://github.com/natdan/RoboxSlicerExtension/tree/Initial-Flow-Code/java/robox-slicer-extension/robox-slicer-flow/src/main/java/com/roboxing/slicerextension/flow)

nebbian, 2017-06-03:
Looks pretty good to me so far.  Keep up the great work!

natdan, 2017-06-04:
@Benraay OK - if you want to start doing something, before we reach working code, you can checkout https://github.com/natdan/RoboxSlicerExtension/tree/Initial-Flow-Code/java/robox-slicer-extension, switch to Initial-Flow-Code branch and commit away. Do feel free to change anything. Consider this as work-in-process until it gets nice rounded shape so we can submit pull request to @nebbian 's branch. 

Currently, class called Controller is responsible for orchestrating work (invoking 'selected' slicer), checking if scripts are supplied and threading result from slicer through them, invoking postProcessing of the selected slicer and script again... Shape of the slicer code is defined in 'Slic3r' and 'DefaultAMCura' classes. Feel free to start populating 'Slic3r' 's postProcessing method - i.e. re-implement 'slic3r_postprocess.pl' script inside.

What is outstanding (from my perspective, for first MVP version) is reading and parsing of the configuration file, control to produce such configuration file and see what common code we can delegate to Slicer (base class). 

I deliberately put 'selected' slicer - as currently it defaults to Slic3r (as only impl) - see Controller's process method. Do let me know if you get stack at anything. I am sorry I didn't do more - decided to finish tiling (walls) on my kitchen as weather was so nice - the work I though I would be doing next weekend - so only grouting is left and hopefully more coding! :D

Benraay, 2017-06-06:
@natdan Ok I have integrated my code in the Initial flow branch had no time to make some tests.
I should work but need some tests.

Also I saw that @nebbian has started making a more generic gcode parsing separating X, Y, Z , E and F values it's maybe a better point to start.

@nebbian you also said that you found that some post processings are not needed to make it work.
Can you tell us which ones ?

nebbian, 2017-06-07:
Nice work @Benraay.

** Required: **
MIN_TRAVEL_DISTANCE is required, otherwise there are occasions where Slic3r travels 0.00001mm or something, the robox post processor rounds this off, and then can't figure out which direction that travel move is in.

Hints are absolutely required.  They must be present, and enough of them to make it obvious to the slicer what's happening.  You must have a hint after a layer change.

Total layer count at the start of the file, layer numbers on each layer change.

** Not required: **
The MIN_EXTRUSION_LENGTH is not required, it works when set to zero.  I spent a lot of time on this code, and now realise that it's useless.

I think you can repeat travel moves, my code removes these, but I think it's not required.

** Unsure **
Putting speed on the same line as the printing code.

Don't print travel moves before a retraction

Good luck :)

Benraay, 2017-06-07:
Ok I will try to make a more simple and more portable post processing with
this informations.
Thanks.
-- 

Benjamin Raaymakers

CONFIDENTIALITY NOTICE:

The contents of this email message and any attachments are intended solely
for the addressee(s) and may contain confidential and/or privileged
information and may be legally protected from disclosure. If you are not
the intended recipient of this message or their agent, or if this message has
been addressed to you in error, please immediately alert the sender by
reply email and thendelete this message and any attachments. If you are not
the intended recipient, you are hereby notified that any use,
dissemination, copying, or storage of this message or its attachments
is strictly
prohibited.

natdan, 2017-06-10:
@Benraay I've realised that I had another commit that I didn't pushed - so I took your code and slightly re-shuffled it around(*). Do stash your changes before pulling and get ready to do some merge conflicts (or instead of just pulling in the latest - do git fetch followed by git rebase - let me know if you need more precise instructions).

(*) changes I've made:

- Your 'demo' branch of if statement in Main class, when input/output files are not supplied is now in Arguments (ensureArgumentsValid method)
- Main class now just delegates to 'Controller' class a class that orchestrates flow. That class is responsible to find out if we have post-processor, scripts and chain them together. Not all is implemented yet, but I'll work on it
- I've refactored a bit your main class of a Slic3r post-processing (your impl/logic is untouched!). Mostly adding try/finally for closing files and such. I'll probably do some more 'fine tuning' the code making it cleaner - but that'll be later.
- Slicer.invoke method now accepts 'result file' argument.
- I envisaged that all the temp files go in print folder next to x.gcode, x_robox.gcode. Currently I've just added 'xxx.gcode.slicer' for slicer output which is then converted (Slicer's class postProcessing) to xxx.gcode (see impl in Arguments)

In general, only code that was victim of my changes is renaming and deleting of temp file at the end of Slic3r class 

Do let me know if I have stepped anywhere on your toes... O: )

Benraay, 2017-06-11:
@natdan ok happy to hear that it is going forward.
and yes I know the 20% thats takes 80% of the time.
I let you go ahead for the user flow and will concentrate on making the postprocessor better

nebbian, 2017-06-13:
Would I be right in assuming that the control application doesn't yet save the .json file for reading by the flow application?  I've tried compiling both branches, and haven't managed to get the flow application to work yet.

natdan, 2017-06-13:
You're right. I almost added it yesterday, but got 'dragged' away to a theatre O: ) (completely forgotten about it). Saving from control app is going to happen quite soon and I'll concentrate testing it and making code more robust (currently when it fails it just fails in some less controllable way).

natdan, 2017-06-13:
@nebbian now yo mentioned two branches - I've merged Initial-Control-Window branchs to Initial-Flow-Code - it is easier when having both at the same time to do saving/loading config file...

natdan, 2017-06-13:
OK - now Initial-Flow-Code should have up-to-date control and flow app which apparently work...
At least when invoked from IDE with parameters ("-o <path-to-file.gcode> <stl-file>" it invokes Slic3r and then post process it. 

Oh, control window only sets slicer at the moment, but I'll add setting pre/post processor scripts, preview and various slicer parameters as well. Now is time to day-dream about all the options we can add to it. :)

nebbian, 2017-06-15:
Thanks for that @natdan.

I've downloaded the latest code, built the jar files, and tried running them.  I can run the control window, and it saves a config file in my home directory under CEL Robox.  Awesome!

However when I try invoking the flow via the command line, it barfs.  Here's a dump:
```
$ cd /Applications/CEL/Common/Cura/
SSEnterprise:Cura ben$ java -jar /Users/ben/Dropbox/Apps/3D/Software/Robox/RoboxSlicerExtension_NatDan/java/robox-slicer-extension/robox-slicer-flow/target/robox-slicer-flow-1.0-SNAPSHOT.jar -o test.gcode
null
java.io.IOException: Cannot run program "CuraEngine_exe": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at com.roboxing.slicerextension.flow.DefaultAMCura.invoke(DefaultAMCura.java:51)
	at com.roboxing.slicerextension.flow.Controller.process(Controller.java:74)
	at com.roboxing.slicerextension.flow.Main.main(Main.java:45)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 3 more

```

Any ideas?  CuraEngine_exe is definitely present.

natdan, 2017-06-15:
_"Any ideas? CuraEngine_exe is definitely present."_

Yes. Maybe we'll need invoking bash to invoke script... I'll see to it tonight.

Benraay, 2017-06-15:
Good news we are almost there, I will make some test tonight also

natdan, 2017-06-15:
It's not that. I put temporary fix - locked AM installation on OSX to /Applications/CEL. It should work now (please try it and let me know - I can react and fix it quite quickly). Real solution is to always run jar from place we'll stick it in AM installation and deduct AM installation dir from there. 
That would mean that when we create run config (Eclipse or Idea for instance) we'll need to somehow pass AM installation path to it, too. Otherwise, jar file is going to sit at the right place and will provide that info just by querying its own location.

natdan, 2017-06-15:
OK - now I've fixed that as well... Jar file must be in /Application/CEL/Common (or in other words 'Common' dir of AM installation. Alternative is to add --am-installation-dir /Application/CEL to the list of args you're invoking jar (or just a main class from IDE). The rest is on code to determine where CuraEngine_exe is and invoke it from there...

And there'll be some changes as it really shouldn't be called CuraEngine_exe but something else (in code so far I've proposed CuraEngine_orig - that's what installer is renaming original CuraEngine currently).

nebbian, 2017-06-30:
Hey @natdan, just checking, the instructions up here are awesome, have they been put into the installation instructions markdown file?  
If not I'll see if I can put them in.  I don't want them to be lost when this issue is closed.

natdan, 2017-06-30:
I am sure I didn't! O: ) - if you can that would be really nice :)

nebbian, 2017-06-30:
Done.  
https://github.com/nebbian/RoboxSlicerExtension/commit/9648b126b3d8372e31f2453493349d218dbe4ae6

Thanks for the amazing work you did here!

closed 2017-06-30 by nebbian