nebbian/RoboxSlicerExtension #9: Absolute extrusion

Benraay, 2017-06-19:
Hi,

I added a absolute to relative values script inside the Slic3r script.
maybe it should be in a parent class because it is meant to be generic for all slicers.
But the parent class "Slicer" is abstract, can I write some generic code there or is this "not the java way of coding" ?

let me know I will move it if it's better.

nebbian, 2017-06-19:
I think we should follow DRY principles as much as possible, which means making everything as generic as possible, and only overriding things that are actually different for each slicer.

I believe that abstract classes can have methods (and code inside those methods), so I'd prefer to put as much code as possible inside the parent "Slicer" class.  It might take some clever thought to work out where the line should be drawn between the parent class and the children.

natdan, 2017-06-19:
Go for it. And let us not being afraid refactoring if we think of a better way! :)

Benraay, 2017-06-19:
ok good news I love DRY code, I will move my code to the parent.

closed 2017-06-23 by nebbian