nebbian/RoboxSlicerExtension #11: gcode not working with last version

Benraay, 2017-06-23:
@nebbian 

I made some tests with the latest perl script from master.
the AM wont postprocess this file.
orig is the one from Slic3r
and gcode is the postprocessed from the perl script

[33446203b4404028.gcode.orig.txt](https://github.com/nebbian/RoboxSlicerExtension/files/1096556/33446203b4404028.gcode.orig.txt)
[33446203b4404028.gcode.txt](https://github.com/nebbian/RoboxSlicerExtension/files/1096555/33446203b4404028.gcode.txt)

nebbian, 2017-06-23:
Hmmm.  It's not generating the hints for some reason.  Note that it was working for me.

I'll look into this.

Benraay, 2017-06-23:
good point that you saw this immediately !

Benraay, 2017-06-23:
I also made a little change on the E regex
E([\-0-9\.]+)\s? 
I need to add the question mark at the end because there is no whitespace at the end of the line !

nebbian, 2017-06-23:
@Benraay I've uploaded a change.  The issue was that you were using a skirt, which by default Automaker doesn't use.  It only uses a brim (but confusingly enough calls it a skirt).  I had set up my Slic3r profiles to use a brim, but no skirt.

Anyway the upshot is that you just have to add one line to the hints and it should all work.

nebbian, 2017-06-23:
btw the E regex should work without mods, there is always a space after the E command in the slic3r output (as far as I can see).  There is no space AFTER postprocessing with our perl script.

Granted, the regex that tests for the space doesn't need to be there, but it should always work with the code that slic3r spits out.

Benraay, 2017-06-23:
Ha I forgot it I had already the same bug with the skirt just forgot to add it again ...

Benraay, 2017-06-23:
Got it , I forgot to remove a call to the perl slicer inside Slic3r advanced options !!
I was parsing the gcode twice !
all looks ok for now thanks for your help

closed 2017-06-23 by nebbian