nebbian/RoboxSlicerExtension #12: Installer and control window

Benraay, 2017-06-23:
@natdan I managed to get it work with the script "CuraEngine" and "install"
work good for development purpose but maybe we need to update your installer script.

@nebbian I made some little changes to the postprocess, added a ? on the speed regex : (F([\-0-9\.]+)\s?)
because when it is on a single line there is no whitespace at the end !
also added a tiny test to put z value only when needed.
and at the end commented out the "if(totalExtrusion > 0){" for the unretraction.
I looked at the original curaEngine code and the unretractions are there !

I got a quite good print !
 
next step try with cura 2.5 !!!

natdan, 2017-06-24:
@Benraay Of course. What you had in mind? I am really hoping to take the latest later on today and see what else is needed to be done. And to finally test it myself - end to end (clean AM, install, and see how it goes).

Also, I am still owing @nebbian ability to open control window on first invocation of our 'flow' and keep it open on all subsequent. Beside that, I was hoping that I'll have a quick look how far is my gcode visualiser and maybe incorporate it as well. I know that in current state it is not robust enough (a bit slow on big resulting gcode files) but with ability to turn it on/off we can include it as 'experimental' feature and it might push me more to do extra optimisation (which I have already thought of)...

nebbian, 2017-06-25:
Hey hey!  It works!  I just used the java stuff to obtain a print.  Excellent work from both of you.

I did find a couple of issues, I'll document them in actual issues here in github.

nebbian, 2017-06-30:
Hey @natdan I just got the following text from Pete at CEL.  He has approved this text to go up in the installer.

> Robox warranty is limited to usage of standard hardware, materials, software and settings as supplied by Robox. Specific design within standard Robox functions reduces the risk of malfunction and terminal damage to Robox hardware. Any use other than uses specifically described in the Robox user manual is entirely at the operators own risk. Although parts of this code and alternative settings have been designed for use with Robox they are in no way approved for use under the Robox warranty. No warranty is given or implied by Robox or by the creators of this code, download, install and use entirely at your own risk. By downloading this code you take responsibility for all faults which may occur with your computer system, software and any attached hardware. We recommend that this code is only used with a fully operational Robox to avoid complicated diagnosis of faults.

24c, 2017-06-30:
Hi, I'm trying to install this on my Mac, and I don't have a JDK on my Mac, just the default Java that comes with Automaker that's within the CEL folder, so I assume I need to download a fresh copy of Java to get this to work.

From Terminal...

 **java -jar robox-extensions-installer.jar
No Java runtime present, requesting install.**

natdan, 2017-06-30:
you can always cheat and use AM's java:
```
$ /Applications/CEL/AutoMaker/java/bin/java -jar robox-extensions-installer.jar
```
given that you've installed AM in /Applications/CEL dir. 

It makes sense creating standalone installers in the future (separate for Mac/Linux and Windows).

24c, 2017-06-30:
Yes AM's Java is in the default location, but I still get this, so I assume installer.jar isn't in the right place..

**$ /Applications/CEL/AutoMaker/java/bin/java -jar robox-extensions-installer.jar
Error: Unable to access jarfile robox-extensions-installer.jar**

natdan, 2017-06-30:
You're close. Where did you download robox-extensinos-installer.jar? You should point to that file - maybe ~/Downloads/robox-extensions-installer.jar?

24c, 2017-06-30:
That's where it is and was, but I downloaded the JDK, and it's done i's stuff. I tried Get Info, and alter Open With, but it wouldn't recognise the Java within the CEL folder, permission denied etc.... 
Thanks for your help, just looking at Automaker now, and checking where stuff has gone. :)

24c, 2017-06-30:
Hi 
I downloaded the JDK, let it do its stuff, and now “slicing is failed” when I ask AM to make
LMAO
Mike

natdan, 2017-07-01:
Huh - I might need more info to help you with that! O: ) Did you fix it?

To revert everything to what it was (before installation) all you need to do is to delete /Applications/CEL/Common/Cura/CuraEngine and rename /Applications/CEL/Common/Cura/CuraEngine_orig back to /Applications/CEL/Common/Cura/CuraEngine. That should make AM behave as before.

As for running it with Slic3r, you need Slic3r installed under /Applications/Slic3r (currently that's the hardcoded path it looks for Slic3r) and you need to setup Slic3r to be 'default' slicer in our control program - which (unhelpfully) we didn't provide jar file for. Alternative is to create file ```~/CEL Robox.slicerextension.config```
and put
```
{"slicer": "Slic3r"}
```
in it! :) Let me know if you need help with that. 

Anyway, current state of development is slightly further away to what we would like it to be - not user friendly enough yet. Please do give us another month or so and it will be far easier to setup and configure (and debug). If you are so keen to give it a go now - be ready to play with shell scriptlets and commands we post here! :)

24c, 2017-07-01:
I fixed it by uninstalling AM and reinstalling. :)

I will try again, but need to get a few models out for somebody, so I’ll have to pass for a few days. I guessed you’d need more information, but you’ve given me some pointers for a fix.
Slic3r was just a .dmg file, and was not installed in Applications, yesterday, but it is now. :o

I will keep plugging away at this, so I apologise in advance for “numpty” mode.

24c, 2017-07-01:
Where does "/CEL Robox.slicerextension.config" exist? somewhere in the CEL folder...??

natdan, 2017-07-02:
Ops: ```~/CEL Robox/.slicerextension.config``` is correct path. If not there this code should just revert back to existing default Cura that is supplied with AM. Apparently it might not have done it (hence error you're seeing). That's something definitively to look at as a bug! :)

24c, 2017-07-02:
Should it not be ~/Documents/CEL Robox/.slicerextension.config. ?

natdan, 2017-07-02:
Do you have directory called ~/Documents/CEL Robox? On my mac it is in ~/CEL Robox.

24c, 2017-07-02:
I have both for some reason. :)

natdan, 2017-07-02:
I suspect only the home dir one is correct.

24c, 2017-07-02:
Anyway, I'm just doing a print, but I have the files in place now in the ~/CEL Robox/ folder, using the following in Terminal

**touch ~/.slicerextension.config
echo "{"slicer": "Slic3r"}" >> ~/.slicerextension.config**

Hopefully, I've done that bit right... :)

natdan, 2017-07-02:
:) That will do just as well.

BTW you could have done it with only one '>' and drop touch:
```
echo "{"slicer": "Slic3r"}" > ~/.slicerextension.config
```
and it would have done the same!

24c, 2017-07-02:
Well that's good to know, like I said I'd be making some "numpty" steps. What's interesting, the first CEL Robox folder was created in November 2016, possibly the first AM install? v2, and maybe the other is v3... I don't know. :) The one is the Documents is the last opened etc, with the latest print jobs in it :?
Thanks for the enforced new learning & responses. In 1hr:34 I'll try again when this print is finished. :)

24c, 2017-07-04:
Sorry natdan this is coming up with "Slicing failed" when I try and make in AM..it's tied to the modded "Cura Engine", because it's not opening "Slic3r"

natdan, 2017-07-04:
I'll add more log statements - but in the mean time if you have file ```~/CEL Robox/robox-slicer-extension.log``` could you, please share it here?

24c, 2017-07-04:
Hi again natdan, I noticed previously it there was a dash or hype added to Slic3r.app, so it became “Slic3r-.app”, so I renamed it, and I am still getting this error. Is it because I have a JDK kit installed on my iMac?

> I'll add more log statements - but in the mean time if you have file ~/CEL Robox/robox-slicer-extension.log could you, please share it here?
> 
09:59:16.953 INFO    [java.util.logging.LogManager$RootLogger log] Started robox slicer extension 
09:59:16.956 FINE    [com.roboxing.slicerextension.flow.Arguments process] Invoked with arguments: 
09:59:16.956 FINE    [com.roboxing.slicerextension.flow.Arguments process]   -v 
09:59:16.957 FINE    [com.roboxing.slicerextension.flow.Arguments process]   -p 
09:59:16.957 FINE    [com.roboxing.slicerextension.flow.Arguments process]   -c 
09:59:16.957 FINE    [com.roboxing.slicerextension.flow.Arguments process]   7bf44453b33147c5.roboxprofile 
09:59:16.958 FINE    [com.roboxing.slicerextension.flow.Arguments process]   -o 
09:59:16.958 FINE    [com.roboxing.slicerextension.flow.Arguments process]   7bf44453b33147c5.gcode 
09:59:16.958 FINE    [com.roboxing.slicerextension.flow.Arguments process]   7bf44453b33147c5-0.stl 
09:59:16.959 FINER   [com.roboxing.slicerextension.flow.Arguments process] Argument : -v 
09:59:16.959 FINER   [com.roboxing.slicerextension.flow.Arguments process] Argument : -p 
09:59:16.959 FINER   [com.roboxing.slicerextension.flow.Arguments process] Argument : -c 
09:59:16.960 FINER   [com.roboxing.slicerextension.flow.Arguments process] Argument : -o 
09:59:16.960 FINER   [com.roboxing.slicerextension.flow.Arguments process] Argument : 7bf44453b33147c5-0.stl 
09:59:16.960 CONFIG  [com.roboxing.slicerextension.flow.Arguments ensureArgumentsValid] Assumed AM installation dir as /Applications/CEL 
09:59:16.966 FINE    [com.roboxing.slicerextension.flow.Controller process] Selecting Slic3r - current dir contains 'PrintJobs'; /Users/mikecrane/CEL Robox/PrintJobs/7bf44453b33147c5 
09:59:16.969 FINE    [com.roboxing.slicerextension.flow.Controller process] Invoking Slic3r 
09:59:16.970 FINE    [com.roboxing.slicerextension.flow.Slic3r invoke] Invoking Slic3r with args:  
09:59:16.970 FINE    [com.roboxing.slicerextension.flow.Slic3r invoke]   /Applications/Slic3r.app/Contents/MacOS/Slic3r 
09:59:16.970 FINE    [com.roboxing.slicerextension.flow.Slic3r invoke]   --gui 
09:59:16.970 FINE    [com.roboxing.slicerextension.flow.Slic3r invoke]   -o 
09:59:16.971 FINE    [com.roboxing.slicerextension.flow.Slic3r invoke]   /Users/mikecrane/CEL Robox/PrintJobs/7bf44453b33147c5/7bf44453b33147c5.slicer.gcode 
09:59:16.971 FINE    [com.roboxing.slicerextension.flow.Slic3r invoke]   /Users/mikecrane/CEL Robox/PrintJobs/7bf44453b33147c5/7bf44453b33147c5-0.stl 
09:59:17.017 SEVERE  [com.roboxing.slicerextension.flow.Main main] Error processing :  
java.io.IOException: Cannot run program "/Applications/Slic3r.app/Contents/MacOS/Slic3r": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at com.roboxing.slicerextension.flow.Slic3r.invoke(Slic3r.java:54)
	at com.roboxing.slicerextension.flow.Controller.process(Controller.java:79)
	at com.roboxing.slicerextension.flow.Main.main(Main.java:67)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:248)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 3 more

licerextension.flow.Controller process] Finished post processing DefaultAMCura 
10:20:23.260 INFO    [com.roboxing.slicerextension.flow.Main main] Finished slicing. 
licing.

natdan, 2017-07-04:
Cool. That's exactly what you said - Slic3r is not in /Applications. Did you get chance to drag it from dmg to /Applications folder yet? And what version of it did you 'install'?

I think if you make sure it is there (/Applications dir) and named Slic3r then it should work. In the mean time (and at, I think, @nebbian request) I'll add selecting Slic3r path to 'configuration' (currently called control) app. And upload app as well as jar so it can be installed with rest of the code. You'll need to give me until tonight, but if it slips it might not be there until Friday/Saturday. But, we are getting there! :)

24c, 2017-07-04:
OK, getting somewhere, it turns out if I show  Package Contents of the Slic3r.app, and go down the folders, "Slic3r" is called"Slic3r-", so I've renamed this, and the Slic3r window opens, like the video, but I get an error. I was using the "torture_little_test.stl"
[image]

24c, 2017-07-04:
so OK'd this and now this...my file format in't "Untitled-hhmmss-ddmmyy", so I will just try another A
[image]
dd model

natdan, 2017-07-04:
What version of Slic3r do you have? Mine is 1.2.9

24c, 2017-07-04:
OK, getting similar errors elsewhere now using a fresh Add Model
[image]
The Prusa one off your links, Slic3r 1.35.5
[image]

natdan, 2017-07-04:
It seems that I'll need fresher version of it to give it a go. @Benraay and @nebbian - what was the version you installed?
@24c - is it possible you to fetch this older version first and give it a go instead of prusa's?

24c, 2017-07-04:
Yes, I will...I see there is a newer version 1.3.0(dev) that is recommended now too.
Just tried 1.2.9 and it's working bud... yippee and well smart guys. (y)
I'll try 1.3.0 too in a moment
[image]

24c, 2017-07-04:
Of course I must point slic3r to the Robox config stuff again! :)

natdan, 2017-07-04:
yes - that's true. Also, if you want to continue using robox in 'standard' way - you should amend ```~/CEL Robox/.slicerextension.config```
to have
```
{"slicer": "DefaultAMCura"}
```
BTW - you can always use TextWrangler (or such) to edit that file! :)

24c, 2017-07-04:
I think the errors I was just getting using 1.2.9 are because the ~Applications/CEL/Common?Slic3r/Config/Robox_slic3r_config_bundle.ini was for the later version...
[image]
Maybe this is the version that the other guys are using?

24c, 2017-07-04:
Just for info guys, I tried to download the other version aka Prusa v1.34.1, but there is only 1.35.1 available. I just tried it as an experiment, and everything works fine, except it seems to hang when slicing after exporting the G-code, and saving the file. It is saved, just won't go to the next step, so if you Close Slicer, you get this error message...
[image]
If you click No you stay in the loop, but Yes lets AM continue with the print.

24c, 2017-07-04:
@natdan the latest Prusa version of Slic3r works fine, you just have to open it first and get past the macOS verification when you drag it into the ~/Applications folder. Then it will open within AM without hiccups. :)
[image]

natdan, 2017-07-04:
@24c it makes perfect sense. Thanks! I would completely missed it. I'm glad you've got around it. Next time it is going to be first question if someone's stuff is not working! :)

24c, 2017-07-05:
@natdan For some odd reason, slicing was failing this morning and the logs pointed to some exception re folder names in the Print Jobs folder IIRC, but to get going I've gone back to OEM Cura, as I need to get a model out.
 Just for info editing the hidden ~/CEL Robox/.slicerextension.config text to **{"slicer": "DefaultAMCura"}** didn't revert back to the default slicer, as Automaker just hung in a grey screen never finishing slicing but heating the bed window. I had to revert the CuraEngine name changes to get it to slice as normal.

natdan, 2017-07-05:
Hm. Something is quite odd in all of it. Could you, please, paste log file in here again?
```~/CEL Robox/robox-slicer-extension.log```

24c, 2017-07-05:
08:10:07.849 INFO    [java.util.logging.LogManager$RootLogger log] Started robox slicer extension 
08:10:07.852 FINE    [com.roboxing.slicerextension.flow.Arguments process] Invoked with arguments: 
08:10:07.853 FINE    [com.roboxing.slicerextension.flow.Arguments process]   -v 
08:10:07.853 FINE    [com.roboxing.slicerextension.flow.Arguments process]   -p 
08:10:07.853 FINE    [com.roboxing.slicerextension.flow.Arguments process]   -c 
08:10:07.853 FINE    [com.roboxing.slicerextension.flow.Arguments process]   Normal.roboxprofile 
08:10:07.854 FINE    [com.roboxing.slicerextension.flow.Arguments process]   -o 
08:10:07.854 FINE    [com.roboxing.slicerextension.flow.Arguments process]   Normal.gcode 
08:10:07.854 FINE    [com.roboxing.slicerextension.flow.Arguments process]   Normal-0.stl 
08:10:07.855 FINE    [com.roboxing.slicerextension.flow.Arguments process]   Normal-1.stl 
08:10:07.855 FINE    [com.roboxing.slicerextension.flow.Arguments process]   Normal-2.stl 
08:10:07.855 FINE    [com.roboxing.slicerextension.flow.Arguments process]   Normal-3.stl 
08:10:07.856 FINE    [com.roboxing.slicerextension.flow.Arguments process]   Normal-4.stl 
08:10:07.856 FINE    [com.roboxing.slicerextension.flow.Arguments process]   Normal-5.stl 
08:10:07.856 FINE    [com.roboxing.slicerextension.flow.Arguments process]   Normal-6.stl 
08:10:07.856 FINER   [com.roboxing.slicerextension.flow.Arguments process] Argument : -v 
08:10:07.857 FINER   [com.roboxing.slicerextension.flow.Arguments process] Argument : -p 
08:10:07.857 FINER   [com.roboxing.slicerextension.flow.Arguments process] Argument : -c 
08:10:07.857 FINER   [com.roboxing.slicerextension.flow.Arguments process] Argument : -o 
08:10:07.858 FINER   [com.roboxing.slicerextension.flow.Arguments process] Argument : Normal-0.stl 
08:10:07.858 FINER   [com.roboxing.slicerextension.flow.Arguments process] Argument : Normal-1.stl 
08:10:07.858 FINER   [com.roboxing.slicerextension.flow.Arguments process] Argument : Normal-2.stl 
08:10:07.859 FINER   [com.roboxing.slicerextension.flow.Arguments process] Argument : Normal-3.stl 
08:10:07.859 FINER   [com.roboxing.slicerextension.flow.Arguments process] Argument : Normal-4.stl 
08:10:07.859 FINER   [com.roboxing.slicerextension.flow.Arguments process] Argument : Normal-5.stl 
08:10:07.859 FINER   [com.roboxing.slicerextension.flow.Arguments process] Argument : Normal-6.stl 
08:10:07.860 CONFIG  [com.roboxing.slicerextension.flow.Arguments ensureArgumentsValid] Assumed AM installation dir as /Applications/CEL 
08:10:07.866 FINE    [com.roboxing.slicerextension.flow.Controller process] Selecting DefaultAMCura - current dir does not contain 'PrintJobs'; /Users/mikecrane/CEL Robox/TimeCostTemp/4686 
08:10:07.868 FINE    [com.roboxing.slicerextension.flow.Controller process] Invoking DefaultAMCura 
08:10:07.869 FINE    [com.roboxing.slicerextension.flow.DefaultAMCura invoke] Invoking default slicer; /Applications/CEL/Common/Cura/CuraEngine_orig 
08:10:07.869 FINEST  [com.roboxing.slicerextension.flow.DefaultAMCura invoke] Replaced Normal.gcode with /Users/mikecrane/CEL Robox/TimeCostTemp/4686/Normal.slicer.gcode 
08:10:07.869 FINE    [com.roboxing.slicerextension.flow.DefaultAMCura invoke] Invoking default Cura slicer with args:  
08:10:07.869 FINE    [com.roboxing.slicerextension.flow.DefaultAMCura invoke]   /Applications/CEL/Common/Cura/CuraEngine_orig 
08:10:07.870 FINE    [com.roboxing.slicerextension.flow.DefaultAMCura invoke]   -v 
08:10:07.870 FINE    [com.roboxing.slicerextension.flow.DefaultAMCura invoke]   -p 
08:10:07.870 FINE    [com.roboxing.slicerextension.flow.DefaultAMCura invoke]   -c 
08:10:07.870 FINE    [com.roboxing.slicerextension.flow.DefaultAMCura invoke]   Normal.roboxprofile 
08:10:07.871 FINE    [com.roboxing.slicerextension.flow.DefaultAMCura invoke]   -o 
08:10:07.871 FINE    [com.roboxing.slicerextension.flow.DefaultAMCura invoke]   /Users/mikecrane/CEL Robox/TimeCostTemp/4686/Normal.slicer.gcode 
08:10:07.871 FINE    [com.roboxing.slicerextension.flow.DefaultAMCura invoke]   Normal-0.stl 
08:10:07.871 FINE    [com.roboxing.slicerextension.flow.DefaultAMCura invoke]   Normal-1.stl 
08:10:07.871 FINE    [com.roboxing.slicerextension.flow.DefaultAMCura invoke]   Normal-2.stl 
08:10:07.872 FINE    [com.roboxing.slicerextension.flow.DefaultAMCura invoke]   Normal-3.stl 
08:10:07.872 FINE    [com.roboxing.slicerextension.flow.DefaultAMCura invoke]   Normal-4.stl 
08:10:07.872 FINE    [com.roboxing.slicerextension.flow.DefaultAMCura invoke]   Normal-5.stl 
08:10:07.872 FINE    [com.roboxing.slicerextension.flow.DefaultAMCura invoke]   Normal-6.stl

24c, 2017-07-05:
That was the one that didn't work AFAIK

natdan, 2017-07-05:
@24c thanks. Looking into those I cannot see any issues. Only thing that's left is for me to try to reproduce it! O: )

24c, 2017-07-05:
OK, well I'm tied to this model for 6hr:15, so you're on your own. Odd it was working OK last night, and not this morning. :)

natdan, 2017-07-05:
Probably different model? Or something else subtly different from then. If you remember what it might be (the difference) do let me know...
BTW thanks for all the testing! You must be quite patient to endure all of this! ;)

nebbian, 2017-07-05:
If you can post the model, or some more clues, then we might be able to replicate it.  Right now it's working for every model that I can throw at my setup, so it's hard to fix.

nebbian, 2017-07-05:
Hang on... you're slicing multiple models on the one plate aren't you?

nebbian, 2017-07-05:
https://github.com/prusa3d/Slic3r/issues/313

24c, 2017-07-05:
@natdan I will revert back later and do some more tests, but I'm happy that it can work as I really need variable layer height, and it'll be interesting to see what else happens with this in the future. I'm a good "numty" tester. :D  @nebbian I can't post the model here, maybe email it to you guys, as it's not mine to chuck around in public. No I'm not, I read about that re Prusa slicer, I made a homogenous STL file AFAIK, but yes the original model was damaged as a DWG file and I did repair it in my CAD & convert back to STL.

nebbian, 2017-07-05:
These lines in your log suggest that you're slicing multiple models:
`08:10:07.858 FINER [com.roboxing.slicerextension.flow.Arguments process] Argument : Normal-0.stl
08:10:07.858 FINER [com.roboxing.slicerextension.flow.Arguments process] Argument : Normal-1.stl
08:10:07.858 FINER [com.roboxing.slicerextension.flow.Arguments process] Argument : Normal-2.stl
08:10:07.859 FINER [com.roboxing.slicerextension.flow.Arguments process] Argument : Normal-3.stl
08:10:07.859 FINER [com.roboxing.slicerextension.flow.Arguments process] Argument : Normal-4.stl
08:10:07.859 FINER [com.roboxing.slicerextension.flow.Arguments process] Argument : Normal-5.stl
08:10:07.859 FINER [com.roboxing.slicerextension.flow.Arguments process] Argument : Normal-6.stl`

natdan, 2017-07-05:
@24c you've got my e-mail - you can always send it there. Only thing is that I cannot do anything right now, nor today :(

24c, 2017-07-05:
That's odd @nebbian I'll take your word for it as not seen 1.stl 2.stl etc in a log file before, so that might explain why the costs etc couldn't be calculated. I inherited the model, and it's got text stencilled through it, I wonder if the middle bits of the "B" & "R" are these extraneous bits...that's the only difference I can see, because R,B,R & B are being printed, which is six isolated middles for this text. :)
No rush or worries either, I'm sure now it is the model, and I think it is these daft graphics the guy has added for branding. :/

natdan, 2017-07-05:
You might have duplicated same model 6 times (seven instances printed at the same time)?

24c, 2017-07-05:
@natdan Unless there are seven instances in the Print Jobs folder! I'll check in a moment, but there is only one model in Automaker, not multiples of this laid onto of each other. I'm daft, but not usually that daft. :)

natdan, 2017-07-05:
LOL - maybe there were seven of them for the log you pasted here. Or somehow model is made of 7 parts (no idea how would that be possible).

24c, 2017-07-05:
LMAO too, it was the middles of the capital B's & R's in his logo cum text he was using for the branding. I selected the part and added it to some others to make an homogeneous part, but he must have not grouped the se characters as part of that, or my CAD import for a DWG format has a bug in how it sees this type of model. Either way, I know what it is now...and I have learnt something else, thanks to you guys ... yet again :D

24c, 2017-07-09:
Hi guys, I seem to get the occasional slicing files fail repeating in a cycle, where Slic3r just briefly flashes, after an AutoMaker Make command, and it's not an additional STL file this time, which does the same thing with the Slic3r PE 1.35. Log file if it means something...

10:19:35.724 CONFIG  [com.roboxing.slicerextension.flow.Arguments ensureArgumentsValid] Assumed AM installation dir as /Applications/CEL 
10:19:35.730 FINE    [com.roboxing.slicerextension.flow.Controller process] Selecting Slic3r - current dir contains 'PrintJobs'; /Users/mikecrane/CEL Robox/PrintJobs/08e9d5199c8249d7 
10:19:35.733 FINE    [com.roboxing.slicerextension.flow.Controller process] Invoking Slic3r 
10:19:35.733 FINE    [com.roboxing.slicerextension.flow.Slic3r invoke] Invoking Slic3r with args:  
10:19:35.734 FINE    [com.roboxing.slicerextension.flow.Slic3r invoke]   /Applications/Slic3r.app/Contents/MacOS/Slic3r 
10:19:35.734 FINE    [com.roboxing.slicerextension.flow.Slic3r invoke]   --gui 
10:19:35.734 FINE    [com.roboxing.slicerextension.flow.Slic3r invoke]   -o 
10:19:35.735 FINE    [com.roboxing.slicerextension.flow.Slic3r invoke]   /Users/mikecrane/CEL Robox/PrintJobs/08e9d5199c8249d7/08e9d5199c8249d7.slicer.gcode 
10:19:35.735 FINE    [com.roboxing.slicerextension.flow.Slic3r invoke]   /Users/mikecrane/CEL Robox/PrintJobs/08e9d5199c8249d7/08e9d5199c8249d7-0.stl 
10:19:40.669 FINE    [com.roboxing.slicerextension.flow.Controller process] Finished slicing with Slic3r 
10:19:40.669 FINE    [com.roboxing.slicerextension.flow.Controller process] Invoking post processing Slic3r 
10:19:40.672 SEVERE  [com.roboxing.slicerextension.flow.Main main] Error processing :  
java.io.FileNotFoundException: 08e9d5199c8249d7.slicer.gcode (No such file or directory)
	at java.io.FileInputStream.open0(Native Method)
	at java.io.FileInputStream.open(FileInputStream.java:195)
	at java.io.FileInputStream.<init>(FileInputStream.java:138)
	at java.util.Scanner.<init>(Scanner.java:635)
	at java.util.Scanner.<init>(Scanner.java:629)
	at com.roboxing.slicerextension.flow.Slicer.postProcess(Slicer.java:78)
	at com.roboxing.slicerextension.flow.Slic3r.postProcess(Slic3r.java:61)
	at com.roboxing.slicerextension.flow.Controller.process(Controller.java:84)
	at com.roboxing.slicerextension.flow.Main.main(Main.java:67)

Parallels desktop is shut down, and Slic3r opens in the macOS fine.

24c, 2017-07-10:
@nebbian You remember that multiple part issue with OSX & Slic3r PE, and the Prusa guy couldn't repeat it in Windows, and suggested it was a wxWidget or Perl related.Have you seen this?
https://github.com/alexrj/Slic3r/wiki/Building-wxWidgets-3.1.0.

nebbian, 2017-07-11:
@24c, it looks like the multiple parts issue has been resolved (just have to wait for the next release): https://github.com/prusa3d/Slic3r/issues/313#issuecomment-314449964

24c, 2017-07-11:
Yes, that's great... and I'll testing it out. Thanks for the update, as I wasn't subscribed to that thread. Fingers crossed, never really wanted to use Windows for this on my iMac. :)

24c, 2017-07-17:
Guys, Prusa Slic3r v1.36 has been added, and just tried it with a simple two part STL files, as well as a "Duplicate" object, and now Automaker is falling over, but at least the Slic3r is opening multipart files

[image]

[image]

[image]

Might be worth having a try, and I'd posted a comment on the Robox forum too.