nebbian/RoboxSlicerExtension #14: Default Cura slicing is broken

nebbian, 2017-06-25:
After choosing the default Cura slicing engine using the control applet, and then doing a slice, I get this error:

```
Jun 25, 2017 8:13:56 AM com.roboxing.slicerextension.flow.Main main
SEVERE: Error processing : 
java.io.IOException: Cannot rename '5277df8f0d024756.slicer.gcode' to '5277df8f0d024756.gcode'
	at com.roboxing.slicerextension.flow.DefaultAMCura.postProcess(DefaultAMCura.java:38)
	at com.roboxing.slicerextension.flow.Controller.process(Controller.java:76)
	at com.roboxing.slicerextension.flow.Main.main(Main.java:58)
```

However, using Slic3r seems to work fine.  Could it be that you don't need to postprocess output from the default engine?

Benraay, 2017-06-25:
Yes probably we don't need to do anything but just call the default curaEngine_exe 
I did not test this, I will take a look maybe tonight

natdan, 2017-06-25:
Actually we would really like to have it as we can then insert custom post processing scripts and visualiser(s). I'll have a look at why it fails! :)

natdan, 2017-06-25:
@nebbian it works now. Problem was in passing all original arguments to Cura - not one with .slicer.gcode. All is fine now. We are ready to try to add 'pre' and 'post' scripts into the flow! :)

nebbian, 2017-06-30:
Sorry Natdan, I totally missed your message there.  Excellent work!

closed 2017-06-30 by nebbian