nebbian/RoboxSlicerExtension #15: Beta testers

nebbian, 2017-06-30:
Hey @natdan and @Benraay, 

Are we ready for beta testers yet?  

For the record I've got no objections :)

natdan, 2017-06-30:
Lets do it. I'll remove all I added to my current AM installation and go through installation process and slicing. I suggest at least someone else does it as well. In the mean time I'll create PR from the branch...

24c, 2017-07-27:
Has anybody used 1.36.0 and tried to do a two colour part?
I was trying to do a back to back comparison vs AutoMaker's default Cura in normal print mode, and I'm not managing it.
[image]

24c, 2017-08-09:
Anybody? I'm still trying to do this with your Slic3r port on Windows and macOS, despite following the right Slic3r ops as far as I can tell...
You double click on the 3D part, an select the STL file and add the extruder1 or 2 in our case. AutoMaker is still ignoring the generated GCode, even though it is OK in Preview after slicing.

[image]
[image]
[image]
[image]
[f7aca7a609c34ce5.zip](https://github.com/nebbian/RoboxSlicerExtension/files/1210988/f7aca7a609c34ce5.zip)

nebbian, 2017-08-09:
I haven't tried to do a 2 colour part yet, and my robox currently has the single material head in it.  There's no reason to try this on Windows, you'd be better off using the mac software to try to get dual colour printing working.

I'd look at the GCode that the default slicer spits out, and then compare with the GCode that Slic3r spits out, when doing a 2 colour part.  I wouldn't be surprised if the GCode only has one colour in it, and the Robox post processor tries to figure out which parts to put in which colour.

24c, 2017-08-09:
OK

from the Slic3r GCode 
; Filament gcode

_M109 S250 T0 ; set temperature and wait for it to be reached
M109 S250 T1 ; set temperature and wait for it to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
T0 ; change extruder
;LAYER:0
G1 Z0.300 F12000.000 ; move to next layer (0)
G1 E-0.50000 F1800.00000 ; retract
G1 X106.125 Y64.619 F12000.000 ; move to first brim point
G1 E0.50000 F1800.00000 ; unretract
G1 F480
G1 X106.166 Y64.413 E0.03234 ; brim
G1 X106.288 Y64.119 E0.04901 ; brim
G1 X106.405 Y63.944 E0.03234 ; brim
G1 X106.946 Y63.403 E0.11780 ; brim
G1 X107.077 Y63.311 E0.02454 ; brim_
....
_G1 X97.683 Y66.371 E0.65537 ; infill
G1 X98.457 Y66.371 E0.12170 ; infill
G1 X100.629 Y68.543 E0.48325 ; infill
G1 X100.629 Y67.770 E0.12170 ; infill
G1 X99.230 Y66.371 E0.31114 ; infill
G1 X100.004 Y66.371 E0.12170 ; infill
G1 X100.870 Y67.237 E0.19265 ; infill
T1 ; change extruder
G1 E-0.50000 F1800.00000 ; retract
G1 X118.000 Y84.700 F12000.000 ; move to first perimeter point
G1 E0.50000 F1800.00000 ; unretract
G1 F480
G1 X108.300 Y84.700 E1.49275 ; perimeter
G1 X108.300 Y65.300 E2.98551 ; perimeter
G1 X127.700 Y65.300 E2.98551 ; perimeter
G1 X127.700 Y84.700 E2.98551 ; perimeter
G1 X118.060 Y84.700 E1.48352 ; perimeter
G1 X117.689 Y84.449 F12000.000 ; move inwards before travel
G1 X118.000 Y84.164 F12000.000 ; move to first perimeter point_

and the robox.gcode

_; End of Pre print gcode
;LAYER:0 height:0.300
G1 F12000 X106.125 Y64.619 Z0.300 
T1 ; Tool Node duration: 181.75755261856784
;TYPE:SKIRT
M109
G1 B1
G1 F480 X106.166 Y64.413 E0.03234
G1 F480 X106.288 Y64.119 E0.04901
G1 F480 X106.405 Y63.944 E0.03234
G1 F480 X106.946 Y63.403 E0.11780
G1 F480 X107.077 Y63.311 E0.02454
G1 F480 X107.414 Y63.166 E0.05656
G1 F480 X107.619 Y63.125 E0.03208
G1 F480 X128.430 Y63.130 E3.20269
G1 F480 X128.584 Y63.165 E0.02429
G1 F480 X128.884 Y63.289 E0.05003
G1 F480 X129.018 Y63.374 E0.02441_
_G1 F480 X129.596 Y63.945 E0.12485_
...
_G1 F3000 X108.914 Y66.826 B0.15 ;  PROC LESS
G1 F3000 X108.914 Y66.318 B0.09 ;  PROC LESS
G1 F3000 X109.479 Y65.753 B0 ;  PROC LESS
;LAYER:2 height:0.700
; Suppressed Tool Node -T start 209.69998725926993 T finish 224.51209489914206 ; Tool Node duration: 14.812107639872197
;TYPE:SKIN
G1 F12000 X92.000 Y84.443 Z0.700 
;TYPE:WALL-OUTER
M109
G1 B1 F150 E0.60000
G1 F3000 X82.557 Y84.443
G1 F3000 X82.557 Y65.557 E1.29173
G1 F3000 X101.443 Y65.557 E1.29173
G1 F3000 X101.443 Y84.443 E1.29173_

and the gcode

 _generated by Slic3r 1.36.2-prusa3d on 2017-08-09 at 08:44:12
      
; Robox profile for AutoMaker
   ; 

; external perimeters extrusion width = 0.40mm
    ; perimeters extrusion width = 0.40mm
   ; infill extrusion width = 0.40mm
   ; solid infill extrusion width = 0.40mm
    ; top infill extrusion width = 0.30mm
   ; first layer extrusion width = 0.60mm
   
; external perimeters extrusion width = 0.40mm
    ; perimeters extrusion width = 0.40mm
   ; infill extrusion width = 0.40mm
   ; solid infill extrusion width = 0.40mm
    ; top infill extrusion width = 0.30mm
   ; first layer extrusion width = 0.60mm
   
M107 ; disable fan
 ; Filament gcode
 
; Filament gcode
 
T0 ; change extruder
  ;Layer count: 12
 ;LAYER:0
G1 E-0.50000
 G0 F12000 X106.125 Y64.619 Z0.3
   G1 E0.50000
 ;TYPE:SKIRT
 G1 F480 X106.166 Y64.413 E0.03234
   G1 F480 X106.288 Y64.119 E0.04901
   G1 F480 X106.405 Y63.944 E0.03234
   G1 F480 X106.946 Y63.403 E0.11780
   G1 F480 X107.077 Y63.311 E0.02454
   G1 F480 X107.414 Y63.166 E0.05656
   G1 F480 X107.619 Y63.125 E0.03208_
...
_   G1 F480 X100.629 Y69.317 E0.12170
   G1 F480 X97.683 Y66.371 E0.65537
   G1 F480 X98.457 Y66.371 E0.12170
   G1 F480 X100.629 Y68.543 E0.48325
   G1 F480 X100.629 Y67.770 E0.12170
   G1 F480 X99.230 Y66.371 E0.31114
   G1 F480 X100.004 Y66.371 E0.12170
   G1 F480 X100.870 Y67.237 E0.19265
   T1 ; change extruder
  G1 E-0.50000
 G0 F12000 X118.000 Y84.700
  G1 E0.50000
 ;TYPE:WALL-OUTER
 G1 F480 X108.300 Y84.700 E1.49275
   G1 F480 X108.300 Y65.300 E2.98551
   G1 F480 X127.700 Y65.300 E2.98551
   G1 F480 X127.700 Y84.700 E2.98551
   G1 F480 X118.060 Y84.700 E1.48352
   G0 F12000 X117.689 Y84.449
  G0 F12000 X118.000 Y84.164
  G1 F480 X108.836 Y84.164 E1.41033
   G1 F480 X108.836 Y65.836 E2.82065
   G1 F480 X127.164 Y65.836 E2.82065
   G1 F480 X127.164 Y84.164 E2.82065
   G1 F480 X118.060 Y84.164 E1.40109
   G1 E-0.50000
 G0 F12000 X110.237 Y83.870
  G1 E0.50000
 ;TYPE:SKIN
 G1 F480 X109.371 Y83.004 E0.19262
   G1 F480 X109.371 Y82.230 E0.12170
   G1 F480 X110.770 Y83.629 E0.31111_

Doesn't it look like the Robox isn't changing the extruder?

nebbian, 2017-08-12:
Hi @24c, try downloading the latest flow java file, and doing a multiple colour print now.

The problem was that our java postprocessor wasn't handling the T0 and T1 code correctly.  I'm doing a multiple colour print now :-)

24c, 2017-08-12:
I had to double click and select Extruder 1 & extruder 2 for each STL part, rather than use default & 1 for example, but it's previewed OK, sliced, and queued, so now I wait and see. Many thanks, as always guys.