nebbian/RoboxSlicerExtension #18: Windows Compatibility

nebbian, 2017-06-30:
Is anyone able to test the windows compatibility of this code?

I'm unable to, there aren't any windows machines in my house.  Even my VMs are way out of date.

Benraay, 2017-06-30:
Good choice having only Mac's in house !!

I have a Virtualbox but have not tried to connect the Robox inside the VM I tried this already with a usb key and it's a bit complicated to forward the USB to the virtual machine.
I can look if the process is ok when installing.

I also have a the PC of my girlfriend but it's an old Windows XP does AM work on XP ?

natdan, 2017-06-30:
I know that the code as it stands won't work on Windows. There are a few issues there for it to work as is. But, I don't mind installing AM on Winodws and add the compatibility for it. I'll set up separate branch and do it as one of the next tasks. Issue (git hub issue - this URL here) is perfect place to record problems, progress, etc...

natdan, 2017-06-30:
Judging by this https://stackoverflow.com/questions/44401911/window-xp-and-java-8 AM might not be able to work on XP. Luckily I have Win2008 and Windows7 - which are perfect low versions of both to test it on (under VMware(*) of course)

(*) - VMware as I happen to have one (3+ years ago I was forced to get one for project we worked on, but VirtualBox should be just fine, too)

Benraay, 2017-06-30:
I have the virtual box for my work also the only point is getting the USB forward to the virtualbox I will make some tests If I get the time I'm busy with my work these days

nebbian, 2017-07-07:
I've gotten the USB connection working.  You need to enable USB 2, and tell VirtualBox to grab the USB connection before the mac does.  Enabling USB 2 requires installing an extension pack.

Screenshot attached.
[image]

24c, 2017-07-07:
@natdan & others, AutoMaker works fine with Windows 8.1 using Parallels 12 on my iMac, and transferred via USB no issues.
[image]

24c, 2017-07-07:
I also created a Slic3r folder in my Program Files folder, and extracted the Prusa Edition into that, cut and pasted the Slic3r folder containing the Robox_slic3r_config_bundle from the iMac into the Common>Cura>Slic3r>Config directory. I then opened Sic3r and did a File>Load Config Bundle op, and everything worked fine.
[image]
Next hurdle is the first line of the @nebbian CuraEngine is not pointing at the right place in Windows, so slicing is failing when you click on Settings from Layout in Automaker, because it can't do any time calls, and that's as far as I can go :)

Benraay, 2017-07-08:
good point for the USB2 tip it was working for me.

I have good news I got it working with the c++ thing now all the parameters are sent to the java flow but I got a new bug 

> Exception in thread "main" java.nio.file.NoSuchFileException: C:\Users\Benjamin-mac\CEL Robox\robox-slicer-extension.log.lck
        at sun.nio.fs.WindowsException.translateToIOException(WindowsException.j
ava:79)
        at sun.nio.fs.WindowsException.rethrowAsIOException(WindowsException.jav
a:97)
        at sun.nio.fs.WindowsException.rethrowAsIOException(WindowsException.jav
a:102)
        at sun.nio.fs.WindowsFileSystemProvider.newFileChannel(WindowsFileSystem
Provider.java:115)
        at java.nio.channels.FileChannel.open(FileChannel.java:287)
        at java.nio.channels.FileChannel.open(FileChannel.java:335)
        at java.util.logging.FileHandler.openFiles(FileHandler.java:459)
        at java.util.logging.FileHandler.<init>(FileHandler.java:292)
        at com.roboxing.slicerextension.flow.Main.main(Main.java:47)

it needs some more investigation look like a path bug  on the user robox folder

Benraay, 2017-07-08:
Ok I found this one, the path to the folders were not the good ones in the flow jar.

But now I have another issue it's freezing when calling the defaultAmCura.
If I look to the task manager CuraEngine_orig.exe is running and CuraEngine.exe also.
it looks normal but the slicing should stop and go to next step but it stays there.

and I need to kill CuraEngine_orig.exe to come back to AM and it says failed.

maybe the way of calling the CuraEngine_orig.exe is not ok on windows ? don't know.

Benraay, 2017-07-08:
Just forgot a thing on the installer on windows we need to copy the "windows%20scripts/CuraEngine.exe" instead of the "shell%20scripts/CuraEngine"

Benraay, 2017-07-08:
also the path to Slic3r needs to be given by people and used in the java code for now it's always calling the mac one

natdan, 2017-07-11:
I am sorry that I couldn't do much last weekend - events took different turn to what I expected so didn't get chance to do much. Anyway - this morning I did something which is going to fit in a gap we have for Windows:
```
using System;

namespace CuraEngine
{
    class Program
    {
        static void Main(string[] args)
        {
            Console.WriteLine("This works and args are:");
            foreach (var arg in args)
            {
                Console.WriteLine("arg=" + arg);
            }
        }
    }
}
```
And result:
```
...\bin\Debug>CuraEngine arg1 -o arg2 arg3
CuraEngin args are:
arg=arg1
arg=-o
arg=arg2
arg=arg3
```
This is quite close. Need to add invoking another executable (AM java with our jar) and we're at last leg of Windows compatibility! :) Maybe today's lunchtime...

natdan, 2017-07-11:
@Benraay  - make another entry (a bug?) here for founds like those. It is much easier to do atomic changes when stressed with time (i.e. me :) )

BTW I have CuraEngine.exe now calling AM's java :)

24c, 2017-07-11:
_BTW I have CuraEngine.exe now calling AM's java :)_
That's good, I'd really like to test the Windows version using multiple parts in Slic3r, as it doesn't fail like macOS apparently. (y)

natdan, 2017-07-11:
OK - I wasted whole evening trying to get my Robox over network working with windows AM. Currently I don't understand why jssc is not called with my 'network serial' port details... Eh.

Anyway, I've committed some windows stuff in 'Windows-Playground' branch. Over there you can find CuraEngine.exe (https://github.com/nebbian/RoboxSlicerExtension/blob/Windows-Playground/windows/CuraEngine/CuraEngine/bin/Debug/CuraEngine.exe) which is supposed to invoke robox-slicer-flow-1.0-SNAPSHOT.jar (see in Program.cs - https://github.com/nebbian/RoboxSlicerExtension/blob/Windows-Playground/windows/CuraEngine/CuraEngine/Program.cs). If that jar is to call correct Slic3r.exe (which I suspect is not) that might be enough for one to bodge all stuff together and give it a go. I am giving up for a day (or two) - until I get chance to look into it more...

24c, 2017-07-11:
@natdan, probably sick of this, but does mean anything...
[image]

24c, 2017-07-11:
OK @natdan got past that one, but I'm failing after importing a STL, this time the Robox robot. When I click on layout, it doesn't calculate the times, and says "Failed"...however it will try and do something if you click on "Make", the printer kicks in, and as expected slicing fails.

24c, 2017-07-11:
almost there, and I can get the original Automaker_exe to run OK, and slice to  print, but not sure I am pointing to the Projects folder on Windows, as it fails hitting Settings still. It hasn't asked me to find this either...I remember seeing a script to do this..sorry if I am being thick.

error details
Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	CuraEngine.exe
  Application Version:	0.0.0.0
  Application Timestamp:	59615bfb
  Fault Module Name:	MSVCP140D.dll
  Fault Module Version:	6.3.9600.18696
  Fault Module Timestamp:	5915ecd6
  Exception Code:	c0000135
  Exception Offset:	0009d4c2
  OS Version:	6.3.9600.2.0.0.256.48
  Locale ID:	2057
  Additional Information 1:	1abe
  Additional Information 2:	1abee00edb3fc1158f9ad6f44f0f6be8
  Additional Information 3:	1abe
  Additional Information 4:	1abee00edb3fc1158f9ad6f44f0f6be8

Read our privacy statement online:
  http://go.microsoft.com/fwlink/?linkid=280262

If the online privacy statement is not available, please read our privacy statement offline:
  C:\Windows\system32\en-GB\erofflps.txt

24c, 2017-07-11:
OK, I pointed to the CEL Robox folder in Windows using Powershell to run the script.. :doh, but I am still getting that MSVCP140D.dll error, with CuraEngine.exe has stopped working...

24c, 2017-07-11:
I'm wondering if I need Visual Studio on my Windows set up, as it doesn't come by default with Parallels 12, it did on previous version but they've messed around with things since they brought out a subscription model a few years back. Downloading now, but I've also reinstalled the original Automaker, as that's what others (google) were saying was the best way to get this dll up and in the system.

24c, 2017-07-12:
Installed both 32 & 64 bit Visual Studio 2015 Runtimes, as suggested by others...diddly squat same error creeping in. Delete or Rename CuraEngine.exe (155kB) version, and no error seen...not sure, time for bed

Benraay, 2017-07-12:
Did the c++ not call the right thing ?
I need a bit time to fix and when I looked to your code I maybe found the bug I probably need to simply exit the progra at the end, 
I will give it a try tonight. 

Also I did'nt use c# because most of the time you need .net framework and other dll's this is a bit heavy for just giving arguments to Java. 

But why not. 

@24c if you want to make some tests coding stuff of corse you need visual studio for compiling the thing

Benraay, 2017-07-12:
Also you can find the c++ source code here 
https://github.com/nebbian/RoboxSlicerExtension/tree/master/windows%20scripts/Source%20C%2B%2B

Maybe it's possible to compile with command line but you need visual studio community edition a least.

natdan, 2017-07-12:
@Benraay doh - I am sorry... I was out of the loop for so long. Sorry for the confusion :(
I suspect your c++ code is, then, the right way forward.

24c, 2017-07-12:
I have tried to install Visual Studio Community on my Parallels Windows 8.1 VM, and it's not liking it for some reason, but as I live in the stick downloads are more problematic and slow here. I will revisit this later, as I need to do a few prints right now. Also tried 2015 x86 64 versions too... :/

Got close to doing something with the .sln, and then it bailed out in VSC 2017.

Benraay, 2017-07-12:
I used VSC 2017 in virtualbox

natdan, 2017-07-12:
I'll delete the branch - my apologies again.

And on that note - I don't think we should commit directly to the master, ever. That's why branches are there and PR. Had I been aware of changes (PRs) I might have not rushed doing what was already done! O: ) (I know - excuses, excuses)

Is it OK for all changes we want to do - just to make a branch and make a PR from the branch? Doesn't matter who is going to approve the PR, but at least we'll have some kind of review process and ability to keep our code clean and/or to get familiar to all changes that go in.

Benraay, 2017-07-12:
Ok sorry for that I will make a branch and PR it to the master when stable

nebbian, 2017-07-12:
Thanks for bringing this up @natdan.  
All changes should be done in their own branch, then a quick pull request and approval.  
Committing directly to master is not really the best way to do things 😀

nebbian, 2017-07-15:
OK I've had a go at getting the windows code to run, and I'm stuck at the first hurdle.  The installer quits after flashing up a message saying that "The system cannot find the path specified".
Any ideas?

nebbian, 2017-07-15:
Never mind, it turns out to be due to Windows not handling the fact that I'm running a file as administrator, from a shared folder.  Gosh.  Copying the folder to C: made it work.  Far out my opinion of Windows hasn't changed much over the years.

Benraay, 2017-07-15:
I switched to mac about 6 years ago and never wanted to go back to windows

Benraay, 2017-07-15:
Sorry It was a bit out of purpose. 

I'm on holidays without printer for two weeks so it's difficult for me to help until I come back. 

When I come back I will try to fix the windows freeze on the arguments passing app

24c, 2017-07-15:
@benraay @nebbian Hopefully by then the macOS multi part Slic3r aka the debugged version will be on the streets. ;)
Off topic, I've used Apple stuff since just after Xerox Star, but I couldn't afford one until the 1990s. Does everything I need, and I when it can't I can use Linux & Windows flavours using Parallels for the obscure engine software and 3D scanner. Definitely getting consistently better quality & control with the Slic3r port, apart from one time in over a dozen+ models.

nebbian, 2017-07-16:
Hey @Benraay, can you describe what you think is happening please?
Also do you have an open source toolset to compile your C++ code?

nebbian, 2017-07-16:
@natdan, I tried to have a look at your Program.cs file, but as the branch is deleted I can't see it.  I'm a bit uncomfortable having to install Microsoft Visual Studio and all of that malarky just to tinker with a simple file that calls a java program.

Benraay, 2017-07-16:
I agree installing Visual Studio is a pain it takes a lot of time and space but when it's done it works very well. 
(I had already Visual studio installed on my virtualbox )
I guess you can compile the code with gcc
But in some tutorials they said you need at least Visual c++ to compile the thing. 
Sorry I did'nt try it. 
But maybe the gcc compiler is the open source one.

natdan, 2017-07-17:
@nebbian I don't mind reinstating branch - but I am not insisting doing anything on it as @Benraay 's is already there. I originally picked C# as it is (supposedly) simpler to do simple things like this than C++ - and easier to understand.

BTW I was just out of luck - finally got some time on Saturday morning, sat on my computer and decided to start long print (4.5h) first and then look at where we stopped. 3 hours later and finally admitting that my dual material head is caput (hopefully only mechanical/electronics - so repairable) I put single material head in started print and went to sofa not feeling well. And spend the rest of the weekend there - not touching coding :( And, now, on Monday, I am fine to go to work! Bah - life is just not fair!

(Branch is re-instated but not for working on and/or creating PR from it)

natdan, 2017-07-17:
@Benraay _"I switched to mac about 6 years ago and never wanted to go back to windows"_ I now it is a bit of a tangent but I am 100% with you there. I've stopped using windows around 2003/4 when my freshly installed Windows (offline) + Northon antivirus (installed offline) and then all updated to the latest as first online stuff, decided to boot (from switched off!) on its own after I shut it down. Ditched all Windows computers in the house, got Mac Mini (only yesterday my son asked me about it - old PPC with 1GB of ram) for household and put Linux on my work laptop. And stayed with such configuration for a while (Linux on my second laptop and went through two more MacMinis - Intel with max 2GB and now one with 8GB - those still upgradable so upgraded HDs now to SSD). In the mean time my wife needed computer so I got her MacAir (5 years later it is still doing well - battery lasts at least 4 hours!) and got my self MBP (first last generation and now first gen with touchbar).

And as it goes in life - needed to use Windows for last 5 years at work (various projects and fortunately only usually 1/3 of the coding time) and at short moment there only VMware worked with that particular Java on windows setup on Mac (Linux was fine - it was some odd tiny bug deep in JBoss or something like that).

nebbian, 2017-07-18:
@Benraay is there any chance that you could write a simple C++ wrapper that then calls a batch file with the same arguments that were passed from AM?  I'm keen to have a crack at sorting this out, but the lack of a compiler is really hurting me.  I did find an .exe online that should call a batch file of the same name, and it worked when called via the CLI, but didn't work when called from AM.

To my mind, our CuraEngine script is simple enough in concept to be handled by a batch file (and this makes updating it a lot easier).

Benraay, 2017-07-18:
yes and it's easy to do with a batch I had it working already, I will try
this tonight but I don't have the printer with me, I will send you the exe
for testing with the printer

Benjamin Raaymakers

CONFIDENTIALITY NOTICE:

The contents of this email message and any attachments are intended solely
for the addressee(s) and may contain confidential and/or privileged
information and may be legally protected from disclosure. If you are not
the intended recipient of this message or their agent, or if this message has
been addressed to you in error, please immediately alert the sender by
reply email and thendelete this message and any attachments. If you are not
the intended recipient, you are hereby notified that any use,
dissemination, copying, or storage of this message or its attachments
is strictly
prohibited.

nebbian, 2017-07-18:
Thanks, much appreciated.

Benraay, 2017-07-18:
Sorry for the master push but I had no time to make a branch.
I hope you will find a way but I have the same "CuraEngine_exe.exe" locked in the  task manager.
maybe something is going wrong when sending parameters need to add some logs around to find out the bug.

you can find in  https://github.com/nebbian/RoboxSlicerExtension/tree/master/windows%20scripts/exe2CmdCall
the exe calling the cmd file

natdan, 2017-07-19:
@Benraay I've got a small request if you don't mind. Is it OK us not to call CuraEngine_exe (or CuraEngine_exe.exe) original batch/executable but CuraEngine_orig/CuraEngine_orig.exe or anything that indicates it is original file?

natdan, 2017-07-19:
@Benraay I've just checked the .cmd file and there's a pause command at the end. I suspect we need something like that but wouldn't that cause issues in the long run?
Also, @nebbian what about stdout from Cura (Cura -> Java -> Batch file -> .exe wrapper)? Do we need to worry about it at all? Does AM use stdout from invoked Cura in any way?

Both: I know we're just experimenting and everything goes until we have stable, running system. These questions are mostly for such potential issues not to be brushed under the carpet.

For instance my understanding is that if batch file invokes Java in separate process and immediately continues with executing the rest of batch - pause will help but you don't want to end up with hundreds of processes stopped at the same pause forever + in such case I cannot see any stdout streaming working at all...

Benraay, 2017-07-19:
@natdan right it is CuraEngine_orig.exe sorry.

the pause is not needed I forgot commenting it out, it was to have error message visible and test purpose.

but you pointed out the thing that maybe goes wrong, i'm not so aware about threads and how processes are working one after the other but something goes wrong here.
I have not read the Robox source code to look how the ask curaEngine.

natdan, 2017-07-19:
@Benraay I am here just to ask questions even though answers may disprove them. :)

It is worth using (sending) CTRL-ALT-DEL and invoking task manager and have a quick peek into what you can see of processes there. When I stupidly invoked itself (in my first versions of CuraEngine) I saw hundreds of processes hanging and no good way of removing them in bulk - needed to restart VM.

BTW Pull requests are helpful for exactly that reason. I'm notoriously bad in forgetting to remove such things from code, leaving out some stupid unfinished tests and such. O: ) Also, they are next best thing after pair programming for sharing code knowledge between people...

Benraay, 2017-07-19:
I totally agree with the pull request I will do it for the next push

nebbian, 2017-07-19:
I've just spent a couple of hours in dll-hell.  The issue here is that the executables produced by @Benraay need some extra dll's to run, which aren't in windows by default.  I've tried adding those DLL's to the Cura folder, but then end up with the error: "The application was unable to start correctly (0xc000007b). Click OK to close the application.".
This occurs with either the .exe that is supposed to call the java file directly, and also with the .exe that is just supposed to call a .cmd file.

I can't imagine how anyone would ever want to code on Windows after going through this.  Gosh.

natdan, 2017-07-19:
LOL +1. I'll have a quick word with someone here if nothing else but regarding .exe file I've produced. I am sure simple 'get parameters and execute another program' shouldn't really rely on many DLL and such code should be embedded in same .exe. I know why it might not have been the case for my .exe - I've just supplied 'debug' version of it without any proper build.

I'll report here what I learn (might not be before Monday when someone who might know more is back in) and hopefully it would be applicable to both C# (for my installer/remote roboxing) and @Benraay 's C++ wrapper...

Benraay, 2017-07-19:
Oops I did the same error it's a debug one 
I will try to remove most dependencies

nebbian, 2017-07-27:
The java file has an issue, it tries to create a log file in C:\Users\[username]\CEL Robox.
This should be C:\Users\[username]\AppData\Roaming\CEL Robox

nebbian, 2017-07-27:
The java flow application is trying to run Cura Engine from C:\Program Files\CEL\CuraEngine_orig.exe, this should be C:\Program Files\CEL\Common\Cura\CuraEngine_orig.exe

nebbian, 2017-07-31:
Gosh.  Windows is just soooo clunky.  I keep getting errors to do with spaces in the command name (ie to do with C:\Program Files\etc), despite having escaped those commands with double quotes.  
Here's where I'm up to now:
* The command to run slic3r is slic3r-console.exe, not slic3r.exe.  This doesn't return until the slic3r window has been closed.
* The slic3r command is currently hard coded to the mac version, instead of selecting a version based on the current OS
* I tried making my own .exe that called slic3r, which almost worked.  The TimeCost values were updating in AutoMaker, and it would open Slic3r when I clicked "Make", but I couldn't get the perl script (that turns the slic3r output into AM compatible Gcode) to run properly, due to Windows not sending through arguments properly to the postprocessing script.
* If I use the java flow code instead, then it works via the command line -- it opens slic3r, you can output the GCode, the flow application reparses it OK and it all looks like it works... except when you open it via AutoMaker.  Then slic3r doesn't open at all.  I'm stumped as to what's going on here.

nebbian, 2017-07-31:
OK I've finally managed to get a print out of AM, using slic3r on Windows 10.  It wasn't easy.

I'll clean up the code a bit and post this in a new branch.  Some testing would be appreciated.

nebbian, 2017-08-05:
OK it's ready for testing the whole kit and kaboodle.  I've merged the windows stuff to the master branch, this should work without too much fiddling around.  
I'd like to see how it goes on a virgin windows system.
I'll open a new issue.

closed 2017-08-05 by nebbian