nebbian/RoboxSlicerExtension #19: Use AM settings

natdan, 2017-07-10:
... to start with when using known slicers (i.e. Slic3r). It would need an controller which would be invoked before invoking a slicer and it will in turn read AM details (from wherever they are stored - probably Cura's format and transfer them to target slicer's config file.

For example: before Slic3r is invoked, read ~/CEL Robox/PrintJobs/<jobid>/<jobid>.roboxprofile and produce slic3r.ini file and then pass that as command line parameter when invoking Slic3r. That way some of the presets from AM would be preserved when Slic3r is inovked. Most importantly material details (what's the point of smart reels if that info is lost in translation).

nebbian, 2017-07-10:
@natdan, just a point of order, the material details ARE preserved (at least the temperatures are, I haven't tested the extrusion multiplier, but I think it would be preserved).  That's part of the magic of using the Robox postprocessor.  It sets the temperatures to whatever is on the smartreel.

I totally agree that we should parse the .roboxprofile and put as much as possible into the command line string for the slicer.  For example, the brim width, layer height, and whether to use support would be prime candidates to get started.

natdan, 2017-07-10:
Actually you are right regarding temperature - it is set in gcode pre and post scripts added to the sliced gcode. Unless temperature (and head speeds!) do have any influence to resulted gcode from selected slicer.

24c, 2017-07-10:
Guys, just out of curiosity, I looked at one of my PrintJob files, and there is a **.gcode**, **.slicer.gcode** & a **.robox.gcode**. I take it the **.robox.gcode** is the version that AM is using in the end? I only ask because my AM temperatures for my filaments are different to the Slic3r ones I imported into Slic3r when I first installed using the config bundle. AM is reporting the AM  temperatures not the Slic3r ones in the window bottom left of AM.
[image]
[image]
[image]

nebbian, 2017-07-10:
@24c, as I mentioned above, the temperatures that get used for printing are the ones that you set in AutoMaker, not Slic3r.  This is done by the postprocessor.

24c, 2017-07-10:
@nebbian, OK, I just looked anyway :)  ...but this is a good thing isn't it?, as you can't damage things.

 I've already seen so far that the support, solid layers etc and infills are all driven by the Slic3r settings, which is pretty cool, as when you Preview the layer, you see how it is going to be made, and these things being implemented. There are a lot of check boxes in Slic3r PE, and I don't know if they all work, as maybe the post processor strips them out, but the infill patterns and support settings are doing fine.