nebbian/RoboxSlicerExtension #29: Windows installer issues

nebbian, 2017-08-03:
After going through the code, it looks like the installer doesn't handle separate files for different OS's.  

At least, I couldn't figure out where it would grab the current CuraEngine.exe from when installing on windows.

Contributing to this is the difficulty in testing an installer that always grabs files from the master branch...

Can someone (@natdan?) let me know how to modify the installer, so that it installs CuraEngine.exe instead of CuraEngine?  And also let me know where to put my updated version of CuraEngine.exe so that it will be installed when someone runs the installer?

Thanks.

natdan, 2017-08-03:
I'll try to do some work tomorrow evening... I might have done it but failed to upload it :(

nebbian, 2017-08-05:
Good stuff @natdan, much appreciated.  I just need confirmation on the CuraEngine.cmd file (ie it's not needed, so it shouldn't be in the installer), and I can then push this branch to master.  This should then let people on Windows try it out :)

closed 2017-08-05 by nebbian