nebbian/RoboxSlicerExtension #30: Windows beta testing

nebbian, 2017-08-05:
I'd appreciate it if anyone who tries to get this working in Windows could post their experiences here.

I've gotten it working with a heavily doctored system, and am wondering how easy it is to get going on a normal windows install.

Try installing using the instructions in the README file, and post here how you go.

24c, 2017-08-05:
OK guys, trying to get started on this for you, but failing at first hurdle...obviously Automaker is in Program files as is Slic3r, but "Robox-extensions-installer" goes where? On the Mac it's in the CEL folder within Applications, but I cannot get it to open using the cmd line...if you're a novice where should we start?

nebbian, 2017-08-05:
Perhaps something like:
"C:\Program Files\CEL\AutoMaker\java\bin\java.exe" -jar robox-extensions-installer.jar
might work?

nebbian, 2017-08-05:
I don't think that it matters where the Robox-extensions-installer goes.  Did you try double clicking the file "Robox-Windows launch Installer here.cmd"?

24c, 2017-08-05:
Yes, and it wants to open with Internet Explorer or Visual Studio :/

24c, 2017-08-05:
but if you double click "Robox-Windows launch installer here" something is happening.

24c, 2017-08-05:
Something has happened, but Automaker is not responding past the Layout page, I cannot add a model, but have drag and dropped an STL file onto the bed, but by going back to the Status tab, I could shift to the Layout page, and then this...
[image]

24c, 2017-08-05:
[image]

24c, 2017-08-05:
Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	CuraEngine.exe
  Application Version:	0.0.0.0
  Application Timestamp:	598016f5
  Fault Module Name:	libmingwex-0.dll
  Fault Module Version:	6.3.9600.18696
  Fault Module Timestamp:	5915ecd6
  Exception Code:	c0000135
  Exception Offset:	0009d4c2
  OS Version:	6.3.9600.2.0.0.256.48
  Locale ID:	2057
  Additional Information 1:	1abe
  Additional Information 2:	1abee00edb3fc1158f9ad6f44f0f6be8
  Additional Information 3:	1abe
  Additional Information 4:	1abee00edb3fc1158f9ad6f44f0f6be8

Read our privacy statement online:
  http://go.microsoft.com/fwlink/?linkid=280262

If the online privacy statement is not available, please read our privacy statement offline:
  C:\Windows\system32\en-GB\erofflps.txt

24c, 2017-08-05:
Is this what the Cura folder should look like or have I got legacy stuff in here?
[image]

nebbian, 2017-08-05:
Gosh.  That's a substantial crash.  What version of windows do you have?

24c, 2017-08-05:
Windows 8.1 on Parallels using the iMac

nebbian, 2017-08-05:
Hmmm.  I'm using Windows 10 on VirtualBox on a mac.

[image]

[image]
[image]
[image]
[image]
[image]

24c, 2017-08-05:
Hmmm, not getting that far, cleaned up the Cura folder... and still got this

Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	CuraEngine.exe.exe
  Application Version:	0.0.0.0
  Application Timestamp:	598016f5
  Fault Module Name:	libmingwex-0.dll
  Fault Module Version:	6.3.9600.18696
  Fault Module Timestamp:	5915ecd6
  Exception Code:	c0000135
  Exception Offset:	0009d4c2
  OS Version:	6.3.9600.2.0.0.256.48
  Locale ID:	2057
  Additional Information 1:	1abe
  Additional Information 2:	1abee00edb3fc1158f9ad6f44f0f6be8
  Additional Information 3:	1abe
  Additional Information 4:	1abee00edb3fc1158f9ad6f44f0f6be8

Read our privacy statement online:
  http://go.microsoft.com/fwlink/?linkid=280262

If the online privacy statement is not available, please read our privacy statement offline:
  C:\Windows\system32\en-GB\erofflps.txt

24c, 2017-08-05:
Does this make any sense,?

https://sourceforge.net/p/mingw/news/2017/02/mingwrt-50-and-w32api-50-snapshots-available/

24c, 2017-08-05:
If I do the usual renaming trick, AutoMaker calculates the print time, and will work as per normal, so there must be an issue somewhere. :/ Where should "Robox-slicer-low-1.0-SNAPSHOT" live?

24c, 2017-08-05:
revert back and problem is still there, time is stalled at "working"...
[image]

nebbian, 2017-08-06:
I've just uploaded a new CuraEngine.exe file, that is statically linked this time.  You can find it here:
https://github.com/nebbian/RoboxSlicerExtension/blob/master/windows%20scripts/CuraEngine.exe
(click "Download").  Give it a go, it appears to work on my Windows 7 VM.

nebbian, 2017-08-06:
@24c:

> If I do the usual renaming trick, AutoMaker calculates the print time, and will work as per normal, so there must be an issue somewhere. :/ Where should "Robox-slicer-flow-1.0-SNAPSHOT" live?

It should be in C:\Program Files\CEL\Common.

24c, 2017-08-06:
OK, I've downloaded the latest version, and double clicked the .cmd file, and everything installed. It doesn't crash now, but neither does it slice either...it just bypasses this mode when you press Make, it queue etc, but doesn't print. I had to rename the slic3r PE, as it had a load of extra characters in the filename, but that still didn't resolve it. Going to download minnow and see if that's what I am doing wrong, but at least it is getting further my iMac running Parallels 12

nebbian, 2017-08-07:
Can you post the following files please?
C:\Users\[YOURUSERNAME]\Documents\CEL Robox\PrintJobs\[LATESTPRINTJOB]\slicer_wrapper.log

C:\Users\[YOURUSERNAME]\Documents\CEL Robox\robox_slicer_extension.log

C:\Users\[YOURUSERNAME]\AppData\Roaming\CEL Robox\AutoMaker.log

This might give us some clues.

Also post a screenshot of windows explorer, showing the contents of C:\Program Files\Slic3r

24c, 2017-08-07:
OK, Ben, first issue, CEL Robox didn’t install into the Documents folder for some reason, but other stuff for info.

> Can you post the following files please?
> C:\Users[YOURUSERNAME]\Documents\CEL Robox\PrintJobs[LATESTPRINTJOB]\slicer_wrapper.log
>
> C:\Users[YOURUSERNAME]\Documents\CEL [...]

Logfile for Robox-slicer-extension

19:16:53.123 INFO    [java.util.logging.LogManager$RootLogger log] Started robox slicer extension 
19:16:53.127 FINE    [com.roboxing.slicerextension.flow.Arguments process] Invoked with arguments: 
19:16:53.128 FINE    [com.roboxing.slicerextension.flow.Arguments process]   -v 
19:16:53.128 FINE    [com.roboxing.slicerextension.flow.Arguments process]   -p 
19:16:53.128 FINE    [com.roboxing.slicerextension.flow.Arguments process]   -c 
19:16:53.130 FINE    [com.roboxing.slicerextension.flow.Arguments process]   ce712db1b4b74193.roboxprofile 
19:16:53.130 FINE    [com.roboxing.slicerextension.flow.Arguments process]   -o 
19:16:53.130 FINE    [com.roboxing.slicerextension.flow.Arguments process]   ce712db1b4b74193.gcode 
19:16:53.130 FINE    [com.roboxing.slicerextension.flow.Arguments process]   ce712db1b4b74193-0.stl 
19:16:53.130 FINER   [com.roboxing.slicerextension.flow.Arguments process] Argument : -v 
19:16:53.131 FINER   [com.roboxing.slicerextension.flow.Arguments process] Argument : -p 
19:16:53.131 FINER   [com.roboxing.slicerextension.flow.Arguments process] Argument : -c 
19:16:53.131 FINER   [com.roboxing.slicerextension.flow.Arguments process] Argument : -o 
19:16:53.132 FINER   [com.roboxing.slicerextension.flow.Arguments process] Argument : ce712db1b4b74193-0.stl 
19:16:53.133 CONFIG  [com.roboxing.slicerextension.flow.Arguments ensureArgumentsValid] Assumed AM installation dir as C:\Program Files\CEL 
19:16:53.139 FINE    [com.roboxing.slicerextension.flow.Controller process] Selecting Slic3r - current dir contains 'PrintJobs'; L:\Documents\CEL Robox\PrintJobs\ce712db1b4b74193 
19:16:53.143 FINE    [com.roboxing.slicerextension.flow.Controller process] Invoking Slic3r 
19:16:53.143 FINE    [com.roboxing.slicerextension.flow.Slic3r invoke] Invoking Slic3r with args:  
19:16:53.143 FINE    [com.roboxing.slicerextension.flow.Slic3r invoke]   /Applications/Slic3r.app/Contents/MacOS/Slic3r 
19:16:53.144 FINE    [com.roboxing.slicerextension.flow.Slic3r invoke]   --gui 
19:16:53.144 FINE    [com.roboxing.slicerextension.flow.Slic3r invoke]   -o 
19:16:53.144 FINE    [com.roboxing.slicerextension.flow.Slic3r invoke]   L:\Documents\CEL Robox\PrintJobs\ce712db1b4b74193\ce712db1b4b74193.slicer.gcode 
19:16:53.144 FINE    [com.roboxing.slicerextension.flow.Slic3r invoke]   L:\Documents\CEL Robox\PrintJobs\ce712db1b4b74193\ce712db1b4b74193-0.stl 
19:16:53.149 SEVERE  [com.roboxing.slicerextension.flow.Main main] Error processing :  
java.io.IOException: Cannot run program "/Applications/Slic3r.app/Contents/MacOS/Slic3r": CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at com.roboxing.slicerextension.flow.Slic3r.invoke(Slic3r.java:54)
	at com.roboxing.slicerextension.flow.Controller.process(Controller.java:79)
	at com.roboxing.slicerextension.flow.Main.main(Main.java:67)
Caused by: java.io.IOException: CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
	at java.lang.ProcessImpl.start(ProcessImpl.java:137)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 3 more

Hope this helps..
Mike

nebbian, 2017-08-07:
Well that will never work!  It's trying to call the mac version of Slic3r.  I wonder what version of the robox slicer extension jarfile you're running?  Perhaps try replacing c:/Program Files/CEL/Common/robox-slicer-flow-1.0-SNAPSHOT.jar with this one:
https://github.com/nebbian/RoboxSlicerExtension/blob/master/robox-slicer-flow-1.0-SNAPSHOT.jar

natdan, 2017-08-07:
Our extension tried to invoke Mac Slic3r not Windows. @nebbian I think we need to update our .jar in github - it might be old version there. If I get time I'll do it - but if I don't post here in next hour - please do maven build and update new version of our jar. It is possible you've done it locally and that's the reason it works for you! :)

natdan, 2017-08-07:
LOL - we typed at the same time! :)

nebbian, 2017-08-07:
@natdan, I updated it yesterday :)

natdan, 2017-08-07:
@24c Installer when run should detect previous installation and allow you to uninstall it and install it again. Maybe you should try that if you haven't already...

24c, 2017-08-07:
> Well that will never work! It's trying to call the mac version of Slic3r. I wonder what version of the robox slicer extension jarfile you're running? Perhaps try replacing c:/Program Files/CEL/Common/robox-slicer-flow-1.0-SNAPSHOT.jar with this one:
> 
> 

Interesting, I don’t know why that is, but when I have just put the CEL Robox file in the Documents folder, something is happening

…and that’s all it was, the CEL Robox folder hadn’t installed into the Documents folder, so is this an error in the installer?

Cheers
Mike

nebbian, 2017-08-07:
@24c is it slicing properly now?  Interested to hear what's going on.  I had to do all manner of trickery to get things to work for me, I've forgotten all the things I did to get it all happening.

From memory I thought that the CEL Robox folder in the Documents was actually created by AutoMaker, I could be wrong though.

24c, 2017-08-07:
> @24c <https://github.com/24c> is it slicing properly now? Interested to hear what's going on. I had to do all manner of trickery to get things to work for me, I've forgotten all the things I did to [...]

Hi again, that’s why I thought I’d have a go. I did uninstall stuff previously, and I used the later update, and it did update @natdan. :)
The problem was definitely the CEL Robox install being in a different place, not sure which is right?

The only other thing that might interest you, is when Slic3r opens as part of the AutoMaker Make sequence, it flashes in the screen briefly, then goes behind AutoMaker window, and flashes in the TaskBar. Obviously clicking on this, brings it to the front…it doesn’t do this in the Mac version

Mike