nebbian/RoboxSlicerExtension #32: Linux support

diego191, 2017-09-01:
I tried to get working your beautiful extension for Robox on Linux. 
I've found that's needed to define path to slic3r in this file: RoboxSlicerExtension/java/robox-slicer-extension/robox-slicer-flow/src/main/java/com/roboxing/slicerextension/flow/OS.java

Then I compiled the robox-slicer-flow-1.0-SNAPSHOT.jar. Finally I run `java -jar robox-extensions-installer.jar` which worked quite well. 
Now after running automaker, importing .stl and pressing "Make" the slic3r (v1.34.1) is opened but it says "_Input file must have .stl, .obj or .amf(.xml) extension_" and it closes slic3r itself.

With slic3r v1.36.2 it says the same error two times and then it says "_Can't call method objects_count on an undefined value at..... Model.pm line 17_" and also error about Multi-part object detected. The slic3r stays opened and I can import STL manually, slice it and export .gcode to folder CEL Robox\PrintJobs\cbe1be0d2ffa4900\cbe1be0d2ffa4900.gcode. However after closing slic3r the Automaker says "Slicing failed".

In the Automaker I can see during slicing this:
```
2017-09-01 13:56:20,666 [Thread-57] ERROR celtech.roboxbase.services.slicer.h.a(SlicerTask.java:353) - Failure when invoking slicer with command line: /home/diego/CEL/AutoMaker/../Common/Cura/CuraEngine -v -p -c cbe1be0d2ffa4900.roboxprofile -o cbe1be0d2ffa4900.gcode cbe1be0d2ffa4900-0.stl
2017-09-01 13:56:20,667 [Thread-57] ERROR celtech.roboxbase.services.slicer.h.a(SlicerTask.java:355) - Slicer terminated with unknown exit code 1
2017-09-01 13:56:21,702 [JavaFX Application Thread] INFO - Print service is in state:SUCCEEDED
```

I tried to run command following command manually and it's processed without error:
```
diego@asterix ~ $ /home/diego/CEL/AutoMaker/../Common/Cura/CuraEngine -v -p -c cbe1be0d2ffa4900.roboxprofile -o cbe1be0d2ffa4900.gcode cbe1be0d2ffa4900-0.stl
14:03:34.400 INFO    [java.util.logging.LogManager$RootLogger log] Started robox slicer extension 
14:03:34.400 INFO    [java.util.logging.LogManager$RootLogger log] Started robox slicer extension 
14:03:34.536 INFO    [com.roboxing.slicerextension.flow.Main main] Finished slicing. 
14:03:34.536 INFO    [com.roboxing.slicerextension.flow.Main main] Finished slicing
```

Do you have an idea how to get it working in Linux? I expect it's very similar to Mac OS. Maybe the problem is in placing script slic3r_postprocess.pl. I cannot find where it should be placed...

24c, 2017-09-01:
diego191 this is the path for macOS
[image]

nebbian, 2017-09-01:
Hi @diego191, welcome aboard!
It sounds like you're really close to finding the solution.  

Quick note, the script slic3r_postprocess.pl is a red herring.  Don't use it.  The post processing is now done in the jar file itself (it used to be done by telling slic3r to use this script to postprocess the gcode).

Regarding the autolaunch of slic3r, I had a similar problem with Windows, believe it or not.  Apparently slic3r takes note of the environment in which it's being launched.  There might be some environmental variables that are not being set properly.  Check the environmental variables for java.  Also check the java log (if there is one).  There should be a log somewhere that logs the RoboxSlicerExtension output.  Post as many logs as you can find, this will really help us to figure out a solution soon.

nebbian, 2017-09-01:
If you just want to marvel at your Robox printing stuff that was produced by Slic3r, you could always open Slic3r, and in the Print Settings tab, enter the path of slic3r_postprocess.pl into the "Output Options" section.

24c, 2017-09-01:
I did read about this.... @diego191 
https://www.reddit.com/r/3Dprinting/comments/6cxttv/slic3r_prusa_edition_wont_open_these_stl_files/

diego191, 2017-09-06:
Hi guys, I'm sorry for the late response. I'm very busy in these days. Thank you for your tips. I just found where is the problem. If I run following command
`slic3r /home/diego/CEL\ Robox/PrintJobs/6b8e4945ab8741a2/6b8e4945ab8741a2-0.stl`
then the slic3r fails.

But if I run this:
```
cd /home/diego/CEL\ Robox/PrintJobs/6b8e4945ab8741a2
slic3r 6b8e4945ab8741a2-0.stl
```
then the model is loade to slic3r correctly! So the problem is in path with white spaces. 
The question is how to fix it? I didn't found any way how to get it work with white spaces in path directly.

natdan, 2017-09-06:
I just cannot catch up with current affairs leave alone this project. I started daydreaming what is needed and how I can help. Anyway - out of curiosity - did you try something like:
```
slic3r "/home/diego/CEL\ Robox/PrintJobs/6b8e4945ab8741a2/6b8e4945ab8741a2-0.stl"
```

Space alone ('\ ') might not be enough...

diego191, 2017-09-06:
Yes, I tried all possible combinations:
```
slic3r /home/diego/CEL\ Robox/PrintJobs/6b8e4945ab8741a2/6b8e4945ab8741a2-0.stl
slic3r "/home/diego/CEL\ Robox/PrintJobs/6b8e4945ab8741a2/6b8e4945ab8741a2-0.stl"
slic3r '/home/diego/CEL\ Robox/PrintJobs/6b8e4945ab8741a2/6b8e4945ab8741a2-0.stl'
slic3r "/home/diego/CEL Robox/PrintJobs/6b8e4945ab8741a2/6b8e4945ab8741a2-0.stl"
slic3r '/home/diego/CEL Robox/PrintJobs/6b8e4945ab8741a2/6b8e4945ab8741a2-0.stl'
slic3r "/home/diego/CEL\\ Robox/PrintJobs/6b8e4945ab8741a2/6b8e4945ab8741a2-0.stl"
...
```

I'm afraid that's bug in slic3r...

natdan, 2017-09-06:
:( - thanks... It was worth trying. I have some other ideas how it might be fixed but can't even count getting near computer until weekend - if then. But this can't go on - I'll get there eventually unless someone beats me to it! :) (hint: process builder has ability to set working directory for subprocess...)