nebbian/RoboxSlicerExtension #35: rbx-toolset

natdan, 2017-11-07:
https://github.com/natdan/rbx-toolset

@Benraay To answer forum quesion here:

And yes – check the link! README over there explains just that. Make a bash script with
```
#!/bin/bash

java -jar /robox-toolset.jar "$@"

```
Call it rbx with right (executable) permission and you can do:
`$ rbx`

For example – the most important commands:
```
$ rbx status
$ rbx -v upload -f xxxx_robox.gcode
$ rbx upload -i myjob < my.gcode
$ rbx pause
$ rbx resume
$ rbx abort
```

But before we use it we’ll need to check if and how much protocol between AM and Robox changed.

Benraay, 2017-11-07:
good news I willl make some tests tonight !!!

natdan, 2017-11-07:
Be careful as I didn't test it since... long time. In theory nothing will go wrong - it will work or not. It is good chance that I didn't implement all the DM stuff fully (or maybe they changed since real release of DM head).

natdan, 2017-11-08:
I've just found something fabulous: https://www.raspberrypi.org/forums/viewtopic.php?f=81&t=196641&p=1232133#p1232133
I'll invest some time to make shell script executable jar files in rbx-toolset project (and all others)...